earthlcd/ezLCD-5x #24: Documentation Woke Issue ezLCD-5xxx Native API Manual.docx

randy-earthlcd, 2022-05-27:
ezLCD-5xxx Native API Manual.docx change name to ezLCD-5x API Manual.docx
Change and replace all 5xxx references to 5x in the manual
Change file name to replace 5xxx with 5x and remove Native from file name

microlan, 2022-06-01:
"Native API" means the raw API, vs Lua API means the Lua API.  Native in computer terms means low-level which is what the RS232 API is.  There is a reason for naming it that way...

microlan, 2022-06-04:
All fixed except "Native" in 060422 or newer docs

randy-earthlcd, 2022-06-05:
Next version add documentation and search for all instances of 'xxx' including removing 4xxx since as far as I'm concerned 4056 RIP! Also the next version should not start with chapter 2. Feel free to close and replace with new issue.