earthlcd/ezLCD-5x #61: ez.I2CWrite(Address, Register, Data [, regSize [, timeout]]) attempt to call a nil value

JacobChrist, 2023-03-16:
**Describe the bug**
Attempting to use: ez.I2CWrite(Address, Register, Data [, regSize [, timeout]])

Called like this:
``` Lua
  result = ez.I2CWrite(42, 0, 0)
```
The following error is in LuaError.log
[string "ezCmdLua"]:824: attempt to call a nil value (field 'I2CWrite')

Going to attempt to stuff value in to a lstringData.

JacobChrist, 2023-03-16:
Same result when trying to stuff into a lstring:

``` Lua
function NAU7802_setRegister(registerAddress, value)
	data = string.char(value)
	result = ez.I2CWrite(_deviceAddress, registerAddress, data, string.len(data))
	return result
end

```

JacobChrist, 2023-03-17:
This is a documentation error, I don't have permissions to update the labels.  The W in Write should not be capitalized in the lua manual.

ez.I2CWrite(Address, Register, Data [, regSize [, timeout]])

should be:

ez.I2Cwrite(Address, Register, Data [, regSize [, timeout]])

microlan, 2023-03-18:
Yes, Lua is case-sensitive.  I confirmed in the source code that the "W" should be lower-case.
Fixed in the manual.  I'll push the update to Git as part of a few other minor docs changes.

closed 2023-03-18 by microlan