earthlcd/ezLCD-5x #65: ezLCD hangs when powered up while enumerating USB

JacobChrist, 2023-03-21:
**Describe the bug**
A clear and concise description of what the bug is.
When the ezLCD-5035 powers up via USB that is connected to a computer that causes the USB-MSD to enumerate this will result in one of two conditions:

1. The Lua script hangs and does not recover.
2. The Lua script  hangs but recovers and continues after enumeration is complete.

**Environment**
- Issue detected via: (Linux file browser open, Lua application running on ezLCD-5035)

**To Reproduce**
Steps to reproduce the behavior:
1. Power ezLCD-5035 via USB that is connected to a computer.

**Expected behavior**
Lua script should not hang or lock up.

microlan, 2023-03-30:
What OS are you using to connect?  Generally customers only use USB for initial programming and not in products.  The issue is that USB and FatFS do not support any kind of locking method so there is no way to control if USB or FatFS "owns" the filesystem at any given time.  Per my research, USB was never designed to support multiple connections to the same file system, so this kind of thing can be a problem if the OS accessing the USB port tries to access the filesystem while the EZLCD code is also accessing FatFS.  The solution everyone came up with online for that was "don't do that--USB was not designed for that"

microlan, 2023-03-30:
NOTE that Windows 10 has no problem with this but Linux may be trying to access the filesystem and possibly trying to set the Dirty bit of the FAT filesystem while we are accessing it in FatFS.  That cannot be supported in the core USB design of MSC unfortunately.

JacobChrist, 2023-03-30:
OS: Linux: 5.15.0-67-generic #74~20.04.1-Ubuntu

It would be a lot of work, but if we could create a fake MSD that the computer talks to that sits between the reals file system and USB this maybe a way around this.  If customers are generally accepting of this issue then it could be closed, however, if I was using this product I would want the ability to not have this occur.  Maybe another option would be to read an input and decide at startup if the device should dual enumerate as MSD + CDC-ACM or just CDC-ACM.

JacobChrist, 2023-04-01:
Started playing with button events in Lua.  Interestingly enough, when the program hangs as USB is enumerating the button events still work (I'm sending data out the serial port in the event).  So the Lua program is not 100% hung.  I'm running ezLCD-5035-032923.dfu.