earthlcd/ezLCD-5x #66: Printing to the screen the font position pointer does not advance

JacobChrist, 2023-03-22:
**Describe the bug**
Sometimes when printing to the screen the font position pointer does not advance resulting in two character being printed at the same screen position.  This happens maybe once every in ten runs.

This maybe related to the USB enumeration issue #65 because that is about where the screen drawing is taking place when the enumeration process starts.

**To Reproduce**
This code is the start of a much larger program, but this is the code that runs up to this point where the issue occurs.

``` Lua

function printLine(font_height, line, str) -- Show a title sequence for the program
	local x1, y1, x2, y2, bg
	-- Display Size -> 320x240 

	-- Erase Old Weight
	x1 = 0
	y1 = font_height * line
	x2 = 320
	y2 = font_height * line + font_height

	bg = (8 * line)

	ez.BoxFill(x1,y1, x2,y2, ez.RGB(bg,bg,bg)) -- X, Y, Width, Height, Color

	-- Display Line
	ez.SetColor(ez.RGB(0,0,255))
	ez.SetFtFont(fn, font_height * 0.70) -- Font Number, Height, Width
	ez.SetXY(x1, y1)
	print(str)
	ez.Wait_ms(1000)
end

function titleScreen(fn) -- Show a title sequence for the program
	ez.Cls(ez.RGB(0,0,0))
	printLine(font_height, 0, "Pull Test - MqpQ")
	ez.Wait_ms(500)
end

fn = 14
font_height = 240 / 8 -- = 30

weight = 0
tare = 0
weight_max = 0
pin = 0

-- Main
titleScreen(fn)

result = ez.I2CopenMaster()
printLine(font_height, 5, "I2C Open: " .. tostring(result) )
ez.Wait_ms(250)

```
**Expected behavior**
The string "I2C Open: true" should have been printed without overlapping characters.

**Screenshots**
Red arrow points to offending error.
[image]

microlan, 2023-03-30:
By design.  We don't support CRLF in the print command because customers almost never start their printing at X=0 so we leave any CRLF-like activities up to the customer.  This matches the behavior of the EZLCD API and the previous 10x Lua code used by existing customers.

closed 2023-03-30 by microlan

JacobChrist, 2023-03-30:
I think you are misunderstanding the issue and this has nothing to do with CRLF.  The same code does different things depending on the run.  Sometimes it works as expected other times the cursor does not advance when printing.

This issue shouldn't be closed.