earthlcd/ezLCD-5x #69: Lua commands to load bitmap files do not work if the file is jpeg format

JacobChrist, 2023-04-03:
**Describe the bug**
Lua commands to load bitmap files do not work if the file is jpeg format.

Tested with both ez.PutPicNo() and ez.PutPictFile().

Also tested trying to load bmp file (which seem to work fine with both functions.

**To Reproduce**
``` Lua
    ez.PutPictFile(0, 0, "Images/EarthLCD_320x240_Splash.jpeg") -- doesn't work
    ez.PutPictFile(0, 0, "/Images/EarthLCD_320x240_Splash.jpeg") -- doesn't work
    ez.PutPictFile(0, 0, "Images/EarthLCD_320x240_Splash.bmp")	-- works
    ez.PutPictFile(0, 0, "/Images/EarthLCD_320x240_Splash.bmp") -- works
```

JacobChrist, 2023-04-03:
Here is the firmware version I'm using for the 5035:
Ver: 3.0.0  Mar 29 2023 (NoSPI)
Lua 5.3.5
S/N: 2c0022000751393334383939
320x240

I confirmed that the gas guage jpg's are working in my autoexec.lua example as well. So it appears that some jpg's render but not others.

Here is my image (jpg) which does not render:
[image]

Here is my bmp & jpg zipped up:
[pulltest-bg.zip](https://github.com/earthlcd/ezLCD-5x/files/11139618/pulltest-bg.zip)

My bmp does render but it doesn't display how I would expect it to.  As you can see the image is offset to the right (and appears to wrap around).  
[image]

The image was generated by opening the jpg in gimp then exporting as a bmp with these settings.
[image]

JacobChrist, 2023-04-03:
At the suggestion Randy tried the following (no X,Y):

``` Lua
	local result = ez.PutPictFile("/Images/EarthLCD_320x240_Splash.jpeg")
	ez.SerialTx("result=".. tostring(result) .. "\r\n", 80, debug_port)
```

Serial output: result=false

microlan, 2023-04-04:
I suspect this issue may be related to the specific internal JPG format of the file you chose.  We have found that the hardware JPEG decoder doesn't seem to work with certain JPG file formats (not really sure why).

JacobChrist, 2023-04-19:
Doing more testing on this tonight and jpeg's written from GIMP are still an issue.

JacobChrist, 2023-04-20:
Addressed in commit: https://github.com/earthlcd/EZLCD-5xxx-Master/commit/a8ee9e0cce128bca59d7cdbf2892b06de01db48e but not yet released.

JacobChrist, 2024-02-05:
@microlan I've pushed this fix to master and I'm closing.

JacobChrist, 2024-02-05:
Addressed (again) in commit: https://github.com/earthlcd/EZLCD-5xxx-Master/commit/5087ce949399b93036bdf4747b2ef4c6c871a00d and master advanced.

closed 2024-02-05 by JacobChrist