earthlcd/ezLCD-5x #74: The ez.SetAlpha() command doesn't work as expected.

JacobChrist, 2023-04-04:
**Describe the bug**
The ez.SetAlpha() command doesn't work as expected.

A clear and concise description of what the bug is.

**To Reproduce*
The following code
``` Lua
	ez.Cls(ez.RGB(0,0,0))

	ez.SetAlpha(255)
	ez.SetXY(44,55)
	result = ez.PutPictFile(2, 3, "/Scale/circle-alpha.bmp")
	ez.SerialTx("result=".. tostring(result) .. "\r\n", 80, debug_port) -- doesn't work
	printLine(font_height, 6, "/Scale/circle-alpha.bmp")
	ez.Wait_ms(2000)

	ez.SetAlpha(128)
	ez.SetXY(11,12)
	result = ez.PutPictFile(2, 3, "/Scale/pulltest-bg.bmp")
	ez.SerialTx("result=".. tostring(result) .. "\r\n", 80, debug_port) -- doesn't work
	printLine(font_height, 6, "/Scale/pulltest-bg.bmp")
	ez.Wait_ms(2000)

	ez.SetAlpha(64)
	ez.SetXY(66,77)
	result = ez.PutPictFile(2, 3, "/Scale/circle-alpha.bmp")
	ez.SerialTx("result=".. tostring(result) .. "\r\n", 80, debug_port) -- doesn't work
	printLine(font_height, 6, "/Scale/circle-alpha.bmp")
	ez.SetAlpha(255)
	ez.Wait_ms(2000)

	ez.SetAlpha(32)
	ez.SetXY(44,55)
	result = ez.PutPictFile(2, 3, "/Scale/pulltest-bg1.bmp")
	ez.SerialTx("result=".. tostring(result) .. "\r\n", 80, debug_port) -- doesn't work
	printLine(font_height, 6, "/Scale/pulltest-bg1.bmp")
	ez.Wait_ms(2000)
```

Produces images like this where you can't really see the image behind it:
[image]
[image]
[image]

**Expected behavior**
Alpha should show background image behind it.

Attached images used with this code:
[Scale.zip](https://github.com/earthlcd/ezLCD-5x/files/11144658/Scale.zip)

microlan, 2023-04-04:
NOTE:  Review the ezLCD-10x documentation to see how the SetAlpha command was designed to work originally.  I have not verified if the current behavior matches the original spec or not.  See [attached ](https://cdn.shopify.com/s/files/1/0073/8351/5183/files/ezLCD_Plus_Series_LUA_API_Manual.pdf?411)

JacobChrist, 2023-04-05:
Cool, when I get back to looking at this I'll check out the original documentation.  Just capturing issues at the moment.